devanlai/dapboot #1: Self bricking when uploading large payload

perillamint, 2017-08-18:
Hello, while I porting DAPBoot on TS100 soldering iron, I discovered this problems:

* When I try to flash large binary, even it is larger then flash size of STM32, it doesn't fail.
* Flashing large enough binary could brick bootloader itself.

To mitigate this problem, I patched DAPBoot like this: https://github.com/perillamint/dapboot/commit/e3a5a633d44b422397f8cb4b01636647ab44d84a

Before making PR with this patch, I want to hear some comments about this patch.

Thanks.

devanlai, 2017-08-18:
Hi perillamint,
Thanks for reporting this issue. I agree that the bootloader should prevent the user from writing too far into flash, especially if could wrap around and overwrite the bootloader.

Your patch looks sound to me. I've done some refactoring so that the flash size is a little better defined across all targets. I've pushed a tentative branch here:
https://github.com/devanlai/dapboot/compare/prevent-flash-overrun

If you want to try it out, as far as I can tell, it should merge cleanly apart from the functionally equivalent changes in `target_flash_program_array()`

perillamint, 2017-08-18:
Thanks for fix. I think your refactoring much looks cleaner then my "quick n dirty" solution. 👍

devanlai, 2017-08-18:
Great, I've merged the fixes to master.

closed 2017-08-18 by devanlai