devanlai/dapboot #2: Option for using HSI as clock source

perillamint, 2017-08-18:
Although using HSI as USB clock source clearly violates USB spec, some Chinese products are using HSI as their clock source. (I guess they skipping HSE because of their additional BOM cost or layout space.)

Would you mind adding an option allow using HSI as clock source like this patch? https://github.com/perillamint/dapboot/commit/f3e98dced709255650149cc0c75439d30ee7ced2 

Obviously, it needs clear warning sign on that option.

devanlai, 2017-08-18:
Done. @cvrtky was also interested in using the bootloader without a crystal, so it seems worth mainlining.

closed 2017-08-18 by devanlai

cvrtky, 2017-08-22:
Hello Devian
Did you manage to get this working? I've been tearing my hair over this. Reason being that in order to set up a USB clock of 48Mhz you wil end up w. System clock of only 6Mhz, which in to low for the APB bus the USB device is connected to. I'll be eager to studdy your code again. Setting the sysclock to 64Mhz before you jump to the actual firmware is simple enough. Either that or requiring the firmware to do it. I've doing the later using you current daploader. I'm still puzeled about how it works w.o. a x-tal. I'v been ill, so haven't had much til looking into that part. It has been interresting studying your code, it is rater well writen.

Best Wishes from Denmark.
Sincerely Tim

Sendt from my iPad

devanlai, 2017-08-22:
Hi Tim,

I have tested the bootloader running from the HSI and it seemed to function fine.

As far as I understand it, apart from the HSI clock tolerance being out of spec, there's nothing in the datasheet that implies that you shouldn't be able to clock everything correctly.

The constraints are:

- The APB1 clock frequency must be >= 8MHz in order to talk to the USB peripheral correctly
- The USB clock needs a clock frequency of 48MHz, fed from the PLL through either a /1 or /1.5 prescaler.

`rcc_clock_setup_in_hsi_out_48mhz()` uses the HSI to run the PLL at 48MHz and it feeds the USB clock with a /1 prescaler, so it has the 48MHz clock that it needs. The APB1 frequency is divided down to keep it below 36MHz, so APB1 is set to 24MHz which is >= 8MHz.

I haven't tried this, but if you want to run the system clock at 8MHz while using USB, you could keep the same PLL and USB prescaler settings as above, but feed the system clock from the HSI and set the APB1 prescaler to 1, so the system clock and APB1 are both running at 8MHz.

cvrtky, 2017-08-24:
Hello Devan
Thank you for your kind answer.
I don't understand how I have missed the option of running the system clock
@ 48 MHz. Probably because I relied to much on clock tree configuration in
the infamous CubeMX. !!!
Understanding the problem and thinking for your self is often more
beneficial than relying on stupid GUI's ;)
I'm not that proficient in libopencm3, but I found the 2 function I will
need to run w.o. X-tal
rcc_clock_setup_in_hsi_out_48mhz()
rcc_clock_setup_in_hsi_out_64mhz()
The later call will come in handy when switching from dapboot to the actual
FW. Should I need a virtual serport, I'll have to do the revers: slow down
to 48 MHz do the IO and the back to full speed :)

Thank you for your time and effort
Sincerely
Tim Børresen - cvrtky@gmail.com