devanlai/dapboot #3: Adding target-specific initialization routine

perillamint, 2017-08-19:
Some hardware, like TS100, has vendor-provided firmware and bootloader pair. That vendor-provided firmware relies on their bootloader do some HW initialization for that firmware.

So, I suggest adding target-specific initialization routine on DAPBoot. It will help when someone wants DAPBoot as alternative bootloader and helps DAPBoot co-exists(boots) their stock firmware without applying hack on their binary builds.

I hacked DAPBoot like this, but I think this patch is much closer to "duct tape" solution rather than elegant one.

https://github.com/perillamint/dapboot/commit/9966fbd641f371386cdb7a19c8ccacbd7eb4daf1
https://github.com/perillamint/dapboot/commit/cf8b3840797e17920f0f69aebe53e0e967c3768c

Could you mainline this feature in your (elegant) way?

devanlai, 2017-08-19:
Support for target-specific application initialization sounds good to me.

There aren't any examples right now, but the intended target directory structure is:
```
stm32f103
|---- target_stm32f103.c
|
|---- ts100
|     |---- config.h
|     |---- target.c
|
|---- generic
      |---- config.h
```

`target_stm32f103.c` holds common functions that are either identical across all stm32f103 targets or that can be easily configured by substituting values from the target-specific `config.h`.

Code fundamentally specific to the ts100 like application hardware initialization would go in `ts100/target.c`. To allow other platforms to build without adding more copies of the function, I would put a weak default implementation of `target_configure_hardware()` in [`dummy.c`](https://github.com/devanlai/dapboot/blob/master/src/dummy.c).

The only thing I'm not sure about is whether there's a better name than `target_configure_hardware(),`, since a lot of the target functions configure hardware one way or another. Maybe something like `target_pre_main()`.

devanlai, 2017-08-20:
This has been integrated in e53f78f as `target_pre_main()`

closed 2017-08-20 by devanlai

perillamint, 2017-08-22:
Thanks. I was being busy few days so I couldn't do anything about this issue.

Anyway, thanks for this feature support. :)