devanlai/dapboot #8: PR of `usbd_register_extra_string` commit from devanlai/libopencm3?

runn1ng, 2017-12-03:
Hello,

would you consider adding PR from your fork of devanlai/libopencm3 to upstream?

It would be helpful to have the `usbd_register_extra_string` in upstream!

I don't want to add that, since it's your code.

(I thought this is on the main repo, but that's actually a github bug that shows forks wrong. :( https://github.com/libopencm3/libopencm3/commit/90be02008e6b651ce737019b6b39ddfdf85eda6b )

devanlai, 2017-12-03:
Hi, there's actually already a PR to upstream usbd_register_extra_string, libopencm3/libopencm3#849. I'm more than happy to let someone else bring it up to libopencm3 standards.

closed 2017-12-03 by devanlai

runn1ng, 2017-12-03:
Thanks, I could not find it!