devanlai/dapboot #11: Windows 7 weird performance. Hangs after download is finished

dazhbog, 2018-12-10:
Hi Devanlai,

Excellent work on dapboot.

We use chrome on Macos and it works flawlessly and on win10 using zadig.

Recently we used stm32duinos tools (wdi-simple.exe method) on a fresh windows 7 to try and make one bat file install vs. installing and configuring Zadig.

Originally the device shows up as an unknown device on windows.

After running the following command, the device shows up correctly.

`wdi-simple.exe --vid 0x1209 --pid 0xdb42 --type XX --name "Test name" --dest "test-dfu"`

Type 0:winusb and 1:libusb was tested. Same results. 

Download works ok, but at the final stage it hangs and the message says 

> Error during final DFU download ControlTransferOut failed NetworkError A transfer error has occurred.

In chrome's console on Win7 it says
```
dfu-util.js:174 Sleeping for 100ms
dfu-util.js:174 Wrote 1024 bytes
dfu-util.js:174 Sent 612 bytes
dfu-util.js:174 Sleeping for 100ms
dfu-util.js:174 Wrote 612 bytes
dfu-util.js:174 Sending empty block  <<hangs here, device does not restart
```

In chrome's console on Mac it says
```
dfu-util.js:174 Sleeping for 100ms
dfu-util.js:174 Wrote 1024 bytes
dfu-util.js:174 Sent 612 bytes
dfu-util.js:174 Sleeping for 100ms
dfu-util.js:174 Wrote 612 bytes
dfu-util.js:174 Sending empty block
dfu-util.js:174 Final DFU status: state=7, status=0
dfu-util.js:174 Ignored reset error
```
Device restarts succesfully on Macos

Any thoughts on what might be going on? Is the bootloader not getting the final message? Is windows not receiving the final DFU status?

Thanks!

devanlai, 2018-12-11:
Hi @dazhbog, I just tested version 1.10 on my desktop running Windows 7 (64-bit) / Chrome 70.0.3538.110.
My final status looks like:
```
dfu-util.js:172 Sleeping for 100ms
dfu-util.js:172 Wrote 1024 bytes
dfu-util.js:172 Sent 868 bytes
dfu-util.js:172 Sleeping for 100ms
dfu-util.js:172 Wrote 868 bytes
dfu-util.js:172 Sending empty block
dfu-util.js:172 Final DFU status: state=7, status=0
```

I got the same results using the auto-installed WinUSB driver and when installing the driver manually using Zadig. Could you try the original/official [dfu-util](http://dfu-util.sourceforge.net/) CLI tool and see if you get different results?
When I use `dfu-util`, I get:
```
Opening DFU capable USB device...
ID 1209:db42
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device
Download        [=========================] 100%        10084 bytes
Download done.
Sent a total of 10084 bytes
state(7) = dfuMANIFEST, status(0) = No error condition is present
unable to read DFU status after completion
```

Note that you can pass in `-v` (multiple times) to get more verbose logging information from `dfu-util`.

dazhbog, 2019-01-17:
Hi @devanlai ,

I have some more data from playing more with it.

First to answer your question, DFU-util 0.8 and 0.9 both were also hanging at the end of the programming. The device was not dettaching nicely.

I installed Zadig and played with different driver options and also read [this guide](https://community.particle.io/t/tutorial-installing-dfu-driver-on-windows-24-feb-2015/3518) where they recommend libusbK for stability. Switching to this suggested driver makes things work just fine. 

So I went back to the `wdi-simple.exe --vid 0x1209 --pid 0xdb42 --type XX --name "Test name" --dest "test-dfu"` command and made the type to 2 where its the libusbK. 

It works now in chrome and with DFU-util, BUT im very suspicious still, so I'll do some more tests and let you know! Especially in older OSs.

Thanks for all the help

dazhbog, 2019-01-17:
Hi @devanlai, one more thing,

When it was originally having issues, one of the boards somehow got bricked and the only way to bring it up was through ST-Link. No idea how that happened, it never occured before, and we flashed hundreds of boards.

Do you have any recommendations on how to lock the bootloader's part of memory? Maybe we could have an RTC register value that tells dapboot to lock/unlock? Or do it like particle, that you need to flash a lock and unlock firmware? Or through ST-Link?

Any experience on this or pointers would be very helpful 😄 

Thanks!

dazhbog, 2019-01-23:
Hi @devanlai, 

Thanks for the response, super helpful! 😄 

Will play with it more and let you know how it goes!

closed 2019-01-23 by dazhbog