devanlai/dapboot #12: CubeMX minimal example not working correctly

NKelias, 2019-04-30:
Hi,

when playing around with the bootloader, I've stumbled upon the following problem:
When uploading code that was generated using CubeMX on a "Black Pill" board, if I do a blinky example like this:
```
HAL_GPIO_TogglePin(GPIOB, GPIO_PIN_12);
HAL_Delay(100);
```
then the LED will remain lit and not flash. If I change the delay to a busy loop like this: 
```
volatile uint32_t i = 900000;
while (i--);
```
then the LED blinks as expected. I assume this has something to do with HAL relying on the Systick!? Any idea as to what could be causing this?

devanlai, 2019-05-01:
As per the discussion on Gitter, it is possible that there is a conflict where the bootloader has configured peripheral registers that the CubeMX HAL expects to still be at their initial reset values. The obvious area of conflict would be the core system clock configuration and the SysTick, though I would expect it to tick at the wrong rate, rather than failing to tick at all.

I would suggest using a debugger that has support for viewing peripheral registers (using a CMSIS-SVD file), placing a breakpoint at the call to `HAL_Delay()`, and comparing the values of all major peripheral registers in the case where it works (without the bootloader, or with a pared down bootloader) vs the case where it does not work with the bootloader.

Assuming it is a peripheral conflict, you could then implement a `target_pre_main()` function to ensure that those peripherals are correctly reset before launching your application.

Depending on the scope of the changes needed, it would also be possible to roll those changes into the core bootloader code, but I don't have CubeMX setup, so it would be a bit difficult for me to test.

NKelias, 2019-05-07:
Turns out it has nothing to do with the bootloader. STM libraries (HAL as well as StdPeriph) hard-code the IRQ vector table address to `0x8000000`. They need to be changed manually to the new flash start address in order to be loaded correctly.  The Parameter name is `NVIC_VectTab_FLASH` in StdPeriph and `FLASH_BASE` in HAL.

closed 2019-06-11 by NKelias