devanlai/dapboot #25: STLINK_HIGH doesn't fit into flash

danielkucera, 2020-06-01:
```
$ make TARGET=STLINK_HIGH V=1
arm-none-eabi-gcc -Os -flto -g -std=gnu11 -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -MD -Wall -Wundef -I../libopencm3/include -DBOOTLOADER_HIGH -DSTM32F1 -DSEMIHOSTING=0 -I. -I./stm32f103/ -I./stm32f103/stlink/ -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -o webusb.o -c webusb.c
arm-none-eabi-gcc -Os -flto -g -std=gnu11 -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -MD -Wall -Wundef -I../libopencm3/include -DBOOTLOADER_HIGH -DSTM32F1 -DSEMIHOSTING=0 -I. -I./stm32f103/ -I./stm32f103/stlink/ -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -o usb_conf.o -c usb_conf.c
arm-none-eabi-gcc -Os -flto -g -std=gnu11 -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -MD -Wall -Wundef -I../libopencm3/include -DBOOTLOADER_HIGH -DSTM32F1 -DSEMIHOSTING=0 -I. -I./stm32f103/ -I./stm32f103/stlink/ -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -o dummy.o -c dummy.c
arm-none-eabi-gcc -Os -flto -g -std=gnu11 -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -MD -Wall -Wundef -I../libopencm3/include -DBOOTLOADER_HIGH -DSTM32F1 -DSEMIHOSTING=0 -I. -I./stm32f103/ -I./stm32f103/stlink/ -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -o dapboot.o -c dapboot.c
arm-none-eabi-gcc -Os -flto -g -std=gnu11 -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -MD -Wall -Wundef -I../libopencm3/include -DBOOTLOADER_HIGH -DSTM32F1 -DSEMIHOSTING=0 -I. -I./stm32f103/ -I./stm32f103/stlink/ -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -o dfu.o -c dfu.c
arm-none-eabi-gcc -Os -flto -g -std=gnu11 -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -MD -Wall -Wundef -I../libopencm3/include -DBOOTLOADER_HIGH -DSTM32F1 -DSEMIHOSTING=0 -I. -I./stm32f103/ -I./stm32f103/stlink/ -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -o winusb.o -c winusb.c
arm-none-eabi-gcc -Os -flto -g -std=gnu11 -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -MD -Wall -Wundef -I../libopencm3/include -DBOOTLOADER_HIGH -DSTM32F1 -DSEMIHOSTING=0 -I. -I./stm32f103/ -I./stm32f103/stlink/ -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -o stm32f103/backup.o -c stm32f103/backup.c
arm-none-eabi-gcc -Os -flto -g -std=gnu11 -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -MD -Wall -Wundef -I../libopencm3/include -DBOOTLOADER_HIGH -DSTM32F1 -DSEMIHOSTING=0 -I. -I./stm32f103/ -I./stm32f103/stlink/ -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -o stm32f103/target_stm32f103.o -c stm32f103/target_stm32f103.c
arm-none-eabi-gcc -flto -Os -g --static -nostartfiles -specs=nano.specs -L../libopencm3/lib -T./stm32f103/stm32f103x8_high.ld -Wl,-Map=dapboot.map -Wl,--gc-sections -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd ./stm32f103/backup.o ./stm32f103/target_stm32f103.o dapboot.o dfu.o dummy.o usb_conf.o webusb.o winusb.o -lopencm3_stm32f1 -Wl,--start-group -lc -lgcc -lnosys -Wl,--end-group -o dapboot.elf
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: dapboot.elf section `.text' will not fit in region `rom'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: region `rom' overflowed by 880 bytes
collect2: error: ld returned 1 exit status
make: *** [rules.mk:191: dapboot.elf] Error 1
```

danielkucera, 2020-06-01:
highboot branch uses start address `0x0800e000` : 
https://github.com/devanlai/dapboot/blob/highboot/src/stm32f103/stm32f103x8.ld#L31

but master uses `0x0800e600`: 
https://github.com/devanlai/dapboot/blob/master/src/stm32f103/stm32f103x8_high.ld#L26

Why?

devanlai, 2020-06-02:
Hi @danielkucera, can you list the full version of the toolchain you're compiling with? I see that it's using 9.2.1 which is nominally the same version as the CI setup is using, but I'd like to eliminate it as a potential source of error if I can. For reference, the CI is using:
```
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
```

Can you rebuild libopencm3 to make sure it's recompiled with the appropriate optimizations, in case it didn't rebuild it after updating the compiler settings.

The master branch is using a more size optimized version of the bootloader contributed by @dmsc which enables LTO and links against newlib-nano instead of newlib to reduce the size. I believe that @dmsc chose fairly tight margins on the space allocated to the high memory bootloader in order to maximize the amount of flash space available to the application.

danielkucera, 2020-06-02:
Hi @devanlai , after `make clean` in libopencm3 it went down but still not enough:
```
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: dapboot.elf section `.text' will not fit in region `rom'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: region `rom' overflowed by 252 bytes
collect2: error: ld returned 1 exit status
make: *** [rules.mk:191: dapboot.elf] Error 1
```

My gcc version:
```
$ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/9.2.1/lto-wrapper
Target: arm-none-eabi
Configured with: ../configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/lib/include' --mandir='/usr/lib/share/man' --infodir='/usr/lib/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --mandir=/usr/share/man --enable-languages=c,c++,lto --enable-multilib --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --enable-tls --build=x86_64-linux-gnu --target=arm-none-eabi --with-system-zlib --with-gnu-as --with-gnu-ld --with-pkgversion=15:9-2019-q4-0ubuntu1 --without-included-gettext --prefix=/usr/lib --infodir=/usr/share/doc/gcc-arm-none-eabi/info --htmldir=/usr/share/doc/gcc-arm-none-eabi/html --pdfdir=/usr/share/doc/gcc-arm-none-eabi/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --disable-libstdc++-v3 --host=x86_64-linux-gnu --with-headers=no --without-newlib --with-multilib-list=rmprofile CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' INHIBIT_LIBC_CFLAGS=-DUSE_TM_CLONE_REGISTRY=0 AR_FOR_TARGET=arm-none-eabi-ar AS_FOR_TARGET=arm-none-eabi-as LD_FOR_TARGET=arm-none-eabi-ld NM_FOR_TARGET=arm-none-eabi-nm OBJDUMP_FOR_TARGET=arm-none-eabi-objdump RANLIB_FOR_TARGET=arm-none-eabi-ranlib READELF_FOR_TARGET=arm-none-eabi-readelf STRIP_FOR_TARGET=arm-none-eabi-strip
Thread model: single
gcc version 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] (15:9-2019-q4-0ubuntu1) 
```

dmsc, 2020-06-02:
Hi @danielkucera !

As @devanlai said, you are using essentially the same toolchain as we are - here it is:
```
gcc version 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 
```

Can you post the dapboot.map file to investigate where the extra size is coming?

Thanks!

danielkucera, 2020-06-02:
Hi. Sure, here.
[dapboot.zip](https://github.com/devanlai/dapboot/files/4718953/dapboot.zip)

dmsc, 2020-06-02:
This is the main difference:

```diff
  .vectors       0x0000000008000000      0x150 /tmp/dapboot.elf.ltrans0.ltrans.o
                 0x0000000008000000                vector_table
 
-.text           0x000000000800e600     0x1980
+.text           0x000000000800e600     0x1ae8
  *(.text*)
  .text          0x000000000800e600     0x14f8 /tmp/dapboot.elf.ltrans0.ltrans.o
                 0x000000000800fa84                reset_handler
  .text.startup  0x000000000800faf8      0x2ec /tmp/dapboot.elf.ltrans0.ltrans.o
- .text.memcpy   0x000000000800fde4       0x1c lib/thumb/v7-m/nofp/libc_nano.a(lib_a-memcpy-stub.o)
+ .text          0x000000000800fde4       0xec lib/thumb/v7-m/nofp/libc_nano.a(lib_a-memcpy.o)
                 0x000000000800fde4                memcpy
- .text          0x000000000800fe00       0x10 lib/thumb/v7-m/nofp/libc_nano.a(lib_a-strlen.o)
-                0x000000000800fe00                strlen
- .text.strncpy  0x000000000800fe10       0x26 lib/thumb/v7-m/nofp/libc_nano.a(lib_a-strncpy.o)
-                0x000000000800fe10                strncpy
-                0x000000000800fe38                . = ALIGN (0x4)
- *fill*         0x000000000800fe36        0x2 
+ .text          0x000000000800fed0       0x5c lib/thumb/v7-m/nofp/libc_nano.a(lib_a-strlen-stub.o)
+                0x000000000800fed0                strlen
+ .text          0x000000000800ff2c       0x74 lib/thumb/v7-m/nofp/libc_nano.a(lib_a-strncpy.o)
+                0x000000000800ff2c                strncpy
+                0x000000000800ffa0                . = ALIGN (0x4)
```

Your nano-libc is much larger than the original:

fn | original | ubuntu
---|-----------|----
memcpy | 0x1c | 0xec
strlen | 0x10 | 0x5c
strncpy | 0x26 | 0x74

Total is 82 bytes in the original versus 444 bytes in your version.

I would file a bug report to the ubuntu tracker, it seems that they miss-compiled the nano-libc; in the mean time, you could use a patch to include your own libc functions, see this: https://github.com/devanlai/dapboot/pull/23/commits/5f160ecae0dc779b79639f1ca46160a177e25623

danielkucera, 2020-06-02:
Which distro are you using? I can compile in container.

danielkucera, 2020-06-02:
even latest debian fails:
```
$ cat Dockerfile 
FROM debian:latest

RUN apt update && apt install -y make gcc-arm-none-eabi

RUN apt install -y python
```
...
```
$ docker run --workdir /src -it -v"$(pwd):/src/" dapboot-builder make -f release.Makefile
...
  BUILD dapboot-maplemini.bin
  BUILD dapboot-stlink.bin
  BUILD dapboot-bluepill-high.bin
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: dapboot.elf section `.text' will not fit in region `rom'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: region `rom' overflowed by 348 bytes
collect2: error: ld returned 1 exit status
make[1]: *** [rules.mk:191: dapboot.elf] Error 1
make: *** [release.Makefile:72: dapboot-bluepill-high.bin] Error 2
```
[dapboot.zip](https://github.com/devanlai/dapboot/files/4719416/dapboot.zip)

devanlai, 2020-06-02:
Hi,

I use a prebuilt copy of the "gnu-rm" toolchain provided by ARM:
https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm

You can see an example of the script I use to download the toolchain for CircleCI here:
https://github.com/devanlai/dapboot/blob/master/util/install-toolchain.sh

You can override the `PREFIX` environment variable / Make variable to explicitly select a toolchain that is not on the system path, e.g.:
```
PREFIX: "~/toolchains/gcc-arm-embedded/bin/arm-none-eabi"
```

dmsc, 2020-06-03:
Hi!

> even latest debian fails:
> 
> ```
> [dapboot.zip](https://github.com/devanlai/dapboot/files/4719416/dapboot.zip)

Yes, same error, the nano-libc is too big, probably configured with -O2 instead of -Os. Also, the Debian package has not been updated in a while.

I'm using the toolchain from ARM, downloaded from here: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

If you don't want to use the pre-compiled package (I use the 64bit Linux build), compiling yourself is easy: download the source, expand and the type:

```
./install-sources.sh --skip_steps=mingw32
./build-prerequisites.sh --skip_steps=mingw32
./build-toolchain.sh --skip_steps=mingw32
```

First two will complete in about a minute, last will compile everything and take a lot more time, and the libraries will be built for 13 different profiles (each arch type and FP combination) and two flavors (normal and nano) each.

The compiled toolchain is relocatable, so you can copy it on any folder and simply symlink the binaries to your PATH.

valoni, 2020-06-20:
Try to compile with GCC 8.3.1 it create smaller file than 9.2.1 / 9.3.1 (i don't know what of reason but different is about 10 kb and more ..)

twelho, 2023-07-19:
If this is still an issue, it should now be fixed by https://github.com/devanlai/dapboot/pull/51.