devanlai/dapboot #26: Unable to use hex file with WebUSB

MohammedBENHADINE, 2020-10-05:
Hey,
First thanks for this great utility. Very useful.
I'm using a BluePill board with a 64Kbytes Flash memory.
I've used the last bootloader release (dapboot-v1.11) and i flashed it using serial adapter.
After turning boot0 to zero , i've successfully detect DFU device ( i ve used the dfu-utils commands to check device).
Once a upload the hex file the device freezes and don't run the application or show itself as a DFU device anymore .Once i turn the boot0 to 1 i can identify back the blue pill back at system memory boot.
Using a bin file i can see the application running once , after a sleep time in the application , i loose the bluepill again (no app running or DFU device). 
My questions : 
1- Is HEX file format handled by the DFU bootloader ?
2- How to switch back and forward between App & Bootloader (without adding a special code in the app ) ?

Regards

devanlai, 2020-10-05:
Hi @MohammedBENHADINE,

1. dapboot does not support hex files. In theory, the DFU protocol doesn't disallow firmware images in a hex format, but it adds complexity to the implementation, especially to support multiple image formats versus just supporting one format for firmware images. For most applications, converting the hex file to a bin file should be sufficient. If it is important to skip over areas without erasing them (for example, to avoid overwriting data stored in flash), you might be better served by a DfuSe bootloader that expects to write to arbitrary non-contiguous addresses.

2. In general, dapboot will prefer to boot your application if the first few bytes of the application firmware look plausible. It will only stay in the bootloader if it detects a configured GPIO / button is held down at boot or if you add special code to write into the RTC backup register to instruct it to switch into the bootloader.

I'm not sure why you're seeing the application only run once. Could you elaborate a bit on what you mean by "sleep time in the application?" Do you mean time spent in sleep mode, or powering the device down and turning it back on again?

MohammedBENHADINE, 2020-10-05:
Thanks for this clear & complete response.
In my application (which is a LoRa Radio Implementation) , just after sending some sensors data by radio , it goes for a sleep time (HAL_PWR_EnterSTANDBYMode) with an RTC alarm to wake it up after some time (10min).

devanlai, 2020-10-05:
Hmm, I have not done any testing with the bootloader in sleep / deep sleep modes, so I'm not sure if anything could be amiss there. Additionally, it is manipulating the RTC backup registers to decide whether to stay in the bootloader or not, so that could cause issues. The relevant code is here, if you want to compare against how your RTC alarm is set:
https://github.com/devanlai/dapboot/blob/master/src/stm32f103/backup.c:
```cpp
void backup_write(enum BackupRegister reg, uint32_t value) {
    rcc_periph_clock_enable(RCC_PWR);
    rcc_periph_clock_enable(RCC_BKP);

    pwr_disable_backup_domain_write_protect();
    RTC_BKP_DR((int)reg*2) = value & 0xFFFFUL;
    RTC_BKP_DR((int)reg*2+1) = (value & 0xFFFF0000UL) >> 16;
    pwr_enable_backup_domain_write_protect();
}
```

MohammedBENHADINE, 2020-10-05:
I think it's definitely the root cause of the error I've got.
I'll look around for other way to wake up the MCU.
Many thanks for your support

closed 2020-10-05 by MohammedBENHADINE