devanlai/dapboot #48: When the bin file download is being done, how I return the device from DFU to UART?

dleung18, 2022-06-10:
I compiled and uploaded the bluepill bootloader to the generic stm32f103c board.
The web page popped up, and I could flash my bin file into the board with no error.
Now, my application wants to see the board as a comport for communication. 
What can I do to get the usb uart port to appear?
Any help is appreciated.

devanlai, 2022-06-10:
Hi @dleung18,
dapboot just handles installing new firmware through USB DFU. Once you've used it to install your firmware, your application is in control and it's up to your code/firmware to act like a USB virtual COM port if that's what you want. If you already have the firmware side working and you want to interact with it from the browser, you may want to look into the [Web Serial API](https://web.dev/serial/).

dleung18, 2022-06-12:
Understood. Will dig it out myself continuously.
Thanks for your help.