devanlai/dapboot #59: SET ADDRESS Error on upload my firmware

wamk9, 2024-03-31:
Hi everyone!

I'm developing a HID Joystick with DFU on STM32F103C8T6 (Bluepill), and I have a question...

I'm fighting with some things, the first is the Internal flash of DFU, without this the dfu-util don't upload anything and get a error, but ok, i used "@Internal Flash   /0x08000000/8*001Kg,56*001Kg" to correct this on USB_INTERFACE_STRING.

After this, i receive this error on try upload my firmware:  Error during special command "SET_ADDRESS" get_status

Any possibility to help me about this? Thanks!

devanlai, 2024-03-31:
Hi @wamk9, dapboot implements the vanilla DFU protocol where the host just uploads or downloads a flat image to a fixed address. It looks like you're trying to use ST's DfuSe protocol which overloads the DFU protocol with additional commands that set the destination address, etc.

If you need that ability to write to different addresses or selectively erase pages, you'll have to find a different bootloader that implements the DfuSe protocol.

If you don't need that functionality and are just trying to get it to work with dfu-util, you can use something like:
```
dfu-util -D nameofmyfirmware.bin
```

wamk9, 2024-04-02:
Hi @devanlai!

Well, I want to write after the bootloader memory address, because i want to keep intact with some info... I go see what i wanna do, Thanks!